wl500g/wl500g #463: Download links for nightly builds are broken

expert3vision, 2015-06-17:
Links at http://asus.vectormm.net/rtn are broken.

I get this error:
    
    XML Parsing Error: not well-formed
    Location: http://asus.vectormm.net/rtn/6220/wl/WL500gpv2-1.9.2.7-rtn-r7070M-ga77fee8.trx
    Line Number 1, Column 5:HDR0

ryzhovau, 2015-06-17:
http://asus.vectormm.net/rtn/
http://wpte.kicks-ass.net/downloads/...ware/Nightlys/
http://vampik.ru/wl500g/rtn/
http://rtn.alegzz.ru/

...and other build bots has no relations with wl500g team. Please, ask bot maintainers about its issues.

lly-dev, 2015-06-18:
Anyway, you have problems with mime-types on your side - .trx is binary.

closed 2015-06-18 by lly-dev

expert3vision, 2015-06-18:
Indeed, Firefox seems to have a problem with it, but Chrome is fine.