wl500g/wl500g #464: WL500W-1.9.2.7-rtn-r7070 - usb not mounted

pult, 2015-08-02:
Starting with firmware "WL500W-1.9.2.7-rtn-r7070M-ga77fee8.trx" my usb is not mounted.
<log>
...ntfs..Mount options: allow_other,nonempty,noatime,fsname=/dev/discs/disca/part1,blkdev,blksize=4096
...
...ntfs..USB storage: ntfs fs at /dev/discs/disca/part1 mounted to /tmp/mnt/disca_1
<\log>
But folder "/tmp/mnt/" is empty.
With the old firmware "WL500W-1.9.2.7-rtn-r7057M-g0521096.trx" everything is fine.

lly-dev, 2015-08-02:
You provide too small info. Please provide full description of volume not mounted properly.
Guess that you use ntfs volume, right?

pult, 2015-08-08:
USB FLASH  16 BG, formated as NTFS.

lly-dev, 2015-08-09:
It's due to ntfs-3g 2015.3.14AR.1 requires FUSE minimal version 7.12, while we has 7.9 at now.
Have to revert a77fee84

closed 2015-08-09 by lly-dev