wl500g/wl500g #469: Build failed

sikmir, 2015-12-31:
Failed to compile the latest with follow build error:
```
configure: creating ./config.status
config.status: creating Makefile
[ ! -d bpalogin ] || make -C bpalogin GLOBAL_OPTIONS="-DUSB_SUPPORT -DQOS -DWPA2_WMM -DASUS_EXT" EXTRACFLAGS="-mips32 -mtune=mips32 -Wno-pointer-sign -mno-branch-likely  -DBCMWPA2 "
make[1]: Entering directory '/home/sikmir/wl500g/gateway/bpalogin'
's#@@BPALOGIN_CONF@@#/usr/etc/bpalogin.conf#' < bpalogin.h.in > bpalogin.h
/bin/sh: s#@@BPALOGIN_CONF@@#/usr/etc/bpalogin.conf#: No such file or directory
Makefile:39: recipe for target 'bpalogin.h' failed
make[1]: *** [bpalogin.h] Error 127
make[1]: Leaving directory '/home/sikmir/wl500g/gateway/bpalogin'
Makefile:1420: recipe for target 'bpalogin' failed
make: *** [bpalogin] Error 2
```

lly-dev, 2016-01-01:
Your build host has no "sed" utility, just install it.

closed 2016-01-01 by lly-dev

sikmir, 2016-01-01:
No, my host has "sed", but autoconf generate Makefile with "SED = ". As workaround I've just edit that Makefile with "SED = sed" and it's OK. But other Makefiles (except bpalogin) have correct "SED = sed".

lly-dev, 2016-01-01:
Where is it? i.e. what "which sed" says?

sikmir, 2016-01-01:
"which sed" -> /usr/bin/sed. For example gateway/quagga/Makefile has correct "SED = sed".

sikmir, 2016-01-01:
autoconf 2.69.

lly-dev, 2016-01-01:
quagga use more recent pregenerated configure script.

Use workaround, wait for commit will be pushed, or patch bpalogin/configure.in & regenerate configure with autoconf.

diff --git a/gateway/bpalogin/configure.in b/gateway/bpalogin/configure.in
--- a/gateway/bpalogin/configure.in
+++ b/gateway/bpalogin/configure.in
@@ -22,6 +22,6 @@ AC_CHECK_FUNCS(select socket strcspn strerror)
 AC_CHECK_LIB(nsl, gethostbyname)
 AC_CHECK_LIB(socket, connect)

-AC_PATH_PROG(CHKCONFIG, chkconfig, , $PATH:/sbin:/usr/sbin:/usr/local/sbin)
-AC_PATH_PROG(SED, sed, , $PATH:/sbin:/usr/sbin:/bin:/usr/bin:/usr/local/bin)
+AC_PATH_PROG(CHKCONFIG, chkconfig, chkconfig)
+AC_PATH_PROG(SED, sed, sed)
 AC_OUTPUT(Makefile)