wl500g/wl500g #472: ISP terminating pppoe connection causes deadlock on ppp0 interface

adanrg, 2016-02-06:
My provider terminates / re-establish connections in every 10080 minutes (1 week). When this happens, the router gets into a state, which can be only fixed with a reboot.

This is the log:
Feb  6 09:29:58 ntp client: Synchronizing time with time.nist.gov...
Feb  6 11:26:57 pppd[380]: LCP terminated by peer (Connect time expired)
Feb  6 11:26:57 pppd[380]: Connect time 10080.0 minutes.
Feb  6 11:26:57 pppd[380]: Sent 2674033330 bytes, received 3082796553 bytes.
Feb  6 11:26:57 dnsmasq[222]: read /etc/hosts - 16 addresses
Feb  6 11:26:57 dnsmasq-dhcp[222]: read /etc/ethers - 13 addresses
Feb  6 11:26:57 PPPoE: Disconnected
Feb  6 11:26:57 dhcp6c[431]: exiting
Feb  6 11:26:57 dhcp6c[431]: sendto: Network is unreachable
Feb  6 11:26:57 dhcp6c[431]: remove address 2xxx:xxxx:xxx:xxx:xxx:xxxx:xxxx:xxxx/64 on br0
Feb  6 11:26:57 dhcp6c[431]: sendto: Network is unreachable
Feb  6 11:26:57 dnsmasq-dhcp[222]: router advertisement on 2xxx:xxx:xxx:xxx::, old prefix for br0
Feb  6 11:26:57 dhcp6c[431]: can't remove address on ppp0: Cannot assign requested address
Feb  6 11:26:57 pppd[380]: Modem hangup
Feb  6 11:26:57 pppd[380]: Connection terminated.
Feb  6 11:26:57 pppd[380]: Connect time 10080.0 minutes.
Feb  6 11:26:57 pppd[380]: Sent 2674033330 bytes, received 3082796553 bytes.
Feb  6 11:27:07 kernel: unregister_netdevice: waiting for ppp0 to become free. Usage count = 1
Feb  6 11:27:17 kernel: unregister_netdevice: waiting for ppp0 to become free. Usage count = 1
Feb  6 11:27:27 kernel: unregister_netdevice: waiting for ppp0 to become free. Usage count = 1

Last message is repeating in every 10 seconds.

Process tree is:
init-+-dnsmasq
     |-dropbear---dropbear---sh---pstree
     |-eapd
     |-httpd
     |-klogd
     |-nas
     |-pppd---ipv6-down
     |-smsd---smsd
     |-syslogd
     |-udhcpc
     |-watchdog
     `-zcip

Process list is:
  PID USER       VSZ STAT COMMAND
    1 admin     1412 S    /sbin/init
    2 admin        0 SW<  [kthreadd]
    3 admin        0 SW<  [ksoftirqd/0]
    4 admin        0 SW<  [events/0]
    5 admin        0 SW<  [khelper]
   27 admin        0 SW<  [kblockd/0]
   63 admin        0 SW   [pdflush]
   64 admin        0 SW   [pdflush]
   65 admin        0 SW<  [kswapd0]
  116 admin        0 SW<  [mtdblockd]
  205 admin     1396 S    syslogd -m 0 -O /tmp/syslog.log -S -D -l 7 -b 1
  208 admin     1396 S    klogd
  209 admin      836 S    eapd
  211 admin     1028 S    nas
  216 admin     1068 S    dropbear -s
  218 admin      984 S    httpd vlan1
  222 nobody     984 S    dnsmasq
  235 admin        0 SW<  [khubd]
  318 admin        0 SW<  [scsi_eh_0]
  319 admin        0 SW<  [usb-storage]
  322 admin        0 SW<  [scsi_eh_1]
  323 admin        0 SW<  [usb-storage]
  330 admin        0 SW<  [scsi_eh_2]
  331 admin        0 SW<  [usb-storage]
  377 admin     1396 S    /sbin/zcip -p /var/run/zcip0.pid vlan1 /tmp/zcip.script
  378 admin     1400 S    /sbin/udhcpc -i vlan1 -p /var/run/udhcpc0.pid -b -O33 -O121 -O249
  380 admin     1496 D    pppd file /tmp/ppp/options.wan0
  382 admin     1432 S    watchdog
  454 admin        0 SW<  [kjournald]
  468 admin     5512 S    smsd
  469 admin     5572 R    smsd
21515 admin        0 Z    [ipv6-down]
21528 admin     1112 R    dropbear -s
21529 admin     1404 S    -sh

Firmware version is:
  1.9.2.7-rtn-r7373M-g265e1c9

themiron, 2016-02-06:
hi, thanks for the report.
does it still happen (ernel: unregister_netdevice: waiting for ppp0 to become free) if turn ipv6 off?

adanrg, 2016-02-06:
Thanks for the quick reply.

If I disable IPv6 through the web UI (IPv6 Connection Type -> DIsabled), I can terminate / reconnect manually without any problems. Doing that with IPv6 enabled (IPv6 Connection Type -> Automatic IPv6) triggers the problem.

Otherwise, IPv6 connection with this firmware is stable and functions flawlessly. So big thumbs up for the good work! :-)

IPv6 is configured in the following way:
  IPv6 Connection Type: Automatic IPv6
  IPv6 Interface: WAN
WAN IPv6 settings
  Get IPv6 automatically? Yes
WAN DNSv6 settings
  Get DNS Server automatically? Yes
LAN IPv6 settings
  Get IPv6 automatically? Yes
  Advertize LAN Prefix? Yes

Process tree with active IPv6 is:
init-+-cron
     |-dhcp6c
     |-dnsmasq
     |-dropbear---dropbear---sh---pstree
     |-eapd
     |-httpd
     |-klogd
     |-nas
     |-pppd
     |-smsd---smsd
     |-syslogd
     |-udhcpc
     |-watchdog
     `-zcip

Process list with active IPv6 is:
  PID USER       VSZ STAT COMMAND
    1 admin     1412 S    /sbin/init
    2 admin        0 SW<  [kthreadd]
    3 admin        0 SW<  [ksoftirqd/0]
    4 admin        0 SW<  [events/0]
    5 admin        0 SW<  [khelper]
   27 admin        0 SW<  [kblockd/0]
   63 admin        0 SW   [pdflush]
   64 admin        0 SW   [pdflush]
   65 admin        0 SW<  [kswapd0]
  116 admin        0 SW<  [mtdblockd]
  205 admin     1396 S    syslogd -m 0 -O /tmp/syslog.log -S -D -l 7 -b 1
  208 admin     1396 S    klogd
  209 admin      836 S    eapd
  211 admin     1028 S    nas
  216 admin     1068 S    dropbear -s
  218 admin      988 S    httpd vlan1
  222 nobody     976 S    dnsmasq
  231 admin        0 SW<  [khubd]
  318 admin        0 SW<  [scsi_eh_0]
  319 admin        0 SW<  [usb-storage]
  325 admin        0 SW<  [scsi_eh_1]
  326 admin        0 SW<  [usb-storage]
  330 admin        0 SW<  [scsi_eh_2]
  331 admin        0 SW<  [usb-storage]
  377 admin     1396 S    /sbin/zcip -p /var/run/zcip0.pid vlan1 /tmp/zcip.script
  378 admin     1400 S    /sbin/udhcpc -i vlan1 -p /var/run/udhcpc0.pid -b -O33 -O121 -O249
  380 admin     1496 S    pppd file /tmp/ppp/options.wan0
  382 admin     1432 S    watchdog
  431 admin     1412 S    /sbin/dhcp6c -D LL -v ppp0
  454 admin        0 SW<  [kjournald]
  468 admin      892 S    cron
  484 admin     5512 S    smsd
  485 admin     5572 S    smsd
  550 admin     1112 R    dropbear -s
  551 admin     1404 S    -sh

I hope this helps.

themiron, 2016-02-06:
so, in other words you do have dhcpv6 connection over pppoe, right? and no other manual ipv6 settings/routes or other changes additional to fw ones via scripts?

adanrg, 2016-02-06:
Yes, that's right. I have some additional ip6tables calls in the post-firewall script, but that's not related to the original issue IMHO.

themiron, 2016-02-06:
please also prove it, just in case

adanrg, 2016-02-06:
post-firewall is:

```
#!/bin/sh

if [ "0.0.0.0" != "$2" ]; then
    /opt/sbin/firewall start
    logger IPv6 firewall rules updated
fi
```

Firewall script is in /opt/sbin, /opt is a mounted USB pendrive.
It's plain simple:
```
#!/bin/sh

WAN_IF='ppp0'
LAN_IF='br0'

start_fw() {
    ip6tables -A FORWARD -i $LAN_IF -o $WAN_IF -m conntrack --ctstate NEW -j ACCEPT

    ip6tables -A FORWARD -i $WAN_IF -o $LAN_IF -p tcp --dport 6881 -m conntrack --ctstate NEW -j ACCEPT
    ip6tables -A FORWARD -i $WAN_IF -o $LAN_IF -p udp --dport 6881 -m conntrack --ctstate NEW -j ACCEPT
    ip6tables -A FORWARD -i $WAN_IF -o $LAN_IF -p tcp --dport  443 -m conntrack --ctstate NEW -j ACCEPT

    ip6tables -A FORWARD -j logdrop
}

stop_fw() {
    ip6tables -A FORWARD -i $LAN_IF -o $WAN_IF -m conntrack --ctstate NEW -j ACCEPT

    ip6tables -D FORWARD -i $WAN_IF -o $LAN_IF -p tcp --dport 6881 -m conntrack --ctstate NEW -j ACCEPT
    ip6tables -D FORWARD -i $WAN_IF -o $LAN_IF -p udp --dport 6881 -m conntrack --ctstate NEW -j ACCEPT
    ip6tables -D FORWARD -i $WAN_IF -o $LAN_IF -p tcp --dport  443 -m conntrack --ctstate NEW -j ACCEPT

    ip6tables -D FORWARD -j logdrop
}

case $1 in
start)
    start_fw
    ;;
stop)
    stop_fw
    ;;
*)
    echo "Usage: firewall [start|stop]"
    ;;
esac
```

themiron, 2016-02-06:
thanks for that, I need additonal output:
$ route -A inet6
$ ip -6 a

adanrg, 2016-02-06:
route -nA inet6
```
Kernel IPv6 routing table
Destination                                 Next Hop                                Flags Metric Ref    Use Iface
2a01:36c:401:226a::/64                      ::                                      UA    256    0        0 ppp0
2a01:36d:401:226a:61dd:5a7:2b23:80ba/128    2a01:36d:401:226a:61dd:5a7:2b23:80ba    UC    0      1      108 br0
2a01:36d:401:226a::/64                      ::                                      U     256    0        0 br0
fe80::/64                                   ::                                      U     256    0        0 eth0
fe80::/64                                   ::                                      U     256    0        0 vlan0
fe80::/64                                   ::                                      U     256    0        0 eth1
fe80::/64                                   ::                                      U     256    0        0 br0
fe80::/64                                   ::                                      U     256    0        0 vlan1
fe80::/64                                   ::                                      U     256    0        0 ppp0
fe80::/10                                   ::                                      U     1      0        0 ppp0
fe80::/10                                   ::                                      U     256    0        0 ppp0
::/0                                        fe80::343a:1c5b:5611:5de1               UGDA  1024   0    28410 ppp0
::/0                                        ::                                      !n    -1     1    87096 lo
::1/128                                     ::                                      Un    0      1        0 lo
2a01:36c:401:226a::/128                     ::                                      Un    0      1        0 lo
2a01:36c:401:226a:a84d:8c25:ba91:7f4b/128   ::                                      Un    0      1      250 lo
2a01:36d:401:226a::/128                     ::                                      Un    0      1        0 lo
2a01:36d:401:226a:21b:fcff:fe45:2c6d/128    ::                                      Un    0      1      470 lo
fe80::/128                                  ::                                      Un    0      1        0 lo
fe80::/128                                  ::                                      Un    0      1        0 lo
fe80::/128                                  ::                                      Un    0      1        0 lo
fe80::/128                                  ::                                      Un    0      1        0 lo
fe80::/128                                  ::                                      Un    0      1        0 lo
fe80::/128                                  ::                                      Un    0      1        0 lo
fe80::21b:fcff:fe45:2c6d/128                ::                                      Un    0      1        0 lo
fe80::21b:fcff:fe45:2c6d/128                ::                                      Un    0      1        0 lo
fe80::21b:fcff:fe45:2c6d/128                ::                                      Un    0      1       53 lo
fe80::21b:fcff:fe45:2c6d/128                ::                                      Un    0      1        0 lo
fe80::21b:fcff:fe45:2c6d/128                ::                                      Un    0      1        0 lo
fe80::a84d:8c25:ba91:7f4b/128               ::                                      Un    0      1        2 lo
ff02::1/128                                 ff02::1                                 UC    0      0     2235 br0
ff00::/8                                    ::                                      U     256    0        0 eth0
ff00::/8                                    ::                                      U     256    0        0 vlan0
ff00::/8                                    ::                                      U     256    0        0 eth1
ff00::/8                                    ::                                      U     256    0        0 br0
ff00::/8                                    ::                                      U     256    0        0 vlan1
ff00::/8                                    ::                                      U     256    0        0 ppp0
::/0                                        ::                                      !n    -1     1    87096 lo
```

ip -6 a
```
1: lo: <LOOPBACK,MULTICAST,UP,LOWER_UP> mtu 16436
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qlen 1000
    inet6 fe80::21b:fcff:fe45:2c6d/64 scope link
       valid_lft forever preferred_lft forever
4: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qlen 1000
    inet6 fe80::21b:fcff:fe45:2c6d/64 scope link
       valid_lft forever preferred_lft forever
5: vlan0@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500
    inet6 fe80::21b:fcff:fe45:2c6d/64 scope link
       valid_lft forever preferred_lft forever
6: vlan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500
    inet6 fe80::21b:fcff:fe45:2c6d/64 scope link
       valid_lft forever preferred_lft forever
7: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500
    inet6 2a01:36d:401:226a:21b:fcff:fe45:2c6d/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::21b:fcff:fe45:2c6d/64 scope link
       valid_lft forever preferred_lft forever
10: ppp0: <POINTOPOINT,MULTICAST,UP,LOWER_UP> mtu 1492 qlen 3
    inet6 2a01:36c:401:226a:a84d:8c25:ba91:7f4b/64 scope global dynamic
       valid_lft 86255sec preferred_lft 14255sec
    inet6 fe80::a84d:8c25:ba91:7f4b/10 scope link
       valid_lft forever preferred_lft forever
```

themiron, 2016-02-09:
fixed with commit fd432c697f079986a6e2f8adf911a57c73d69e64
r7396+ build fw images will be availabe tomorrow at http://asus.vectormm.net/rtn/

closed 2016-02-09 by themiron