wl500g/wl500g #477: unregister_netdevice tun0 failed

ondrej-bartos, 2016-03-26:
Looks the problem #457 is not resolved:

`Sat Mar 26 11:24:19 2016 SENT CONTROL [techon.cz]: 'PUSH_REQUEST' (status=1)
Sat Mar 26 11:24:19 2016 PUSH: Received control message: 'PUSH_REPLY,route 192.168.1.0 255.255.255.224,route 192.168.4.0 255.255.255.224,topology subnet,tun-mtu 1200,route-gateway 192.168.3.1,topology subnet,ping 30,ping-restart 120,ifconfig 192.168.3.2 255.255.255.0'
Sat Mar 26 11:24:19 2016 Options error: option 'tun-mtu' cannot be used in this context ([PUSH-OPTIONS])
Sat Mar 26 11:24:19 2016 OPTIONS IMPORT: timers and/or timeouts modified
Sat Mar 26 11:24:19 2016 OPTIONS IMPORT: --ifconfig/up options modified
Sat Mar 26 11:24:19 2016 OPTIONS IMPORT: route options modified
Sat Mar 26 11:24:19 2016 OPTIONS IMPORT: route-related options modified
Sat Mar 26 11:24:19 2016 TUN/TAP device tun0 opened
Sat Mar 26 11:24:19 2016 TUN/TAP TX queue length set to 100
Sat Mar 26 11:24:19 2016 do_ifconfig, tt->ipv6=0, tt->did_ifconfig_ipv6_setup=0
Sat Mar 26 11:24:19 2016 /sbin/ifconfig tun0 192.168.3.2 netmask 255.255.255.0 mtu 1200 broadcast 192.168.3.255
Sat Mar 26 11:24:19 2016 /sbin/route add -net 192.168.1.0 netmask 255.255.255.224 gw 192.168.3.1
Sat Mar 26 11:24:19 2016 /sbin/route add -net 192.168.4.0 netmask 255.255.255.224 gw 192.168.3.1
Sat Mar 26 11:24:19 2016 Initialization Sequence Completed

^C

Sat Mar 26 11:41:44 2016 event_wait : Interrupted system call (code=4)
Sat Mar 26 11:41:44 2016 /sbin/route del -net 192.168.4.0 netmask 255.255.255.224
Sat Mar 26 11:41:44 2016 /sbin/route del -net 192.168.1.0 netmask 255.255.255.224
Sat Mar 26 11:41:44 2016 Closing TUN/TAP interface
Sat Mar 26 11:41:44 2016 /sbin/ifconfig tun0 0.0.0.0`

syslog:

tail /tmp/syslog.log Mar 26 11:46:06 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:46:17 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:46:27 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:46:37 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:46:47 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:46:57 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:47:07 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:47:17 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:47:27 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1 Mar 26 11:47:37 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1

uname -a
Linux asus 2.6.22.21 #2 Sat Mar 26 02:25:34 MSK 2016 mips GNU/Linux

Using latest: WL500gpv2-1.9.2.7-rtn-r7438M-g373b73c.trx

themiron, 2016-03-26:
have you enabled ipv6?

ondrej-bartos, 2016-03-26:
No, I did not touch IPv6 at all. 
IPv6 Connection Type: Disabled

ondrej-bartos, 2016-03-26:
[image]

byte02, 2016-04-29:
Same here.
Plus: 
p-t-p openvpn pings ok both sides but no tcp/udp trafic
firewall full open
openvpn ok at [open/dd]-wrt but there are no koi-8 in samba 8]
old RTNs versions lost their googlecode repositories(
stuck with that
thanks in advance

ondrej-bartos, 2016-07-28:
anything new here, how about to try compilation with latest 2.6.x kernel? 2.6.3x?

dvglab, 2016-08-11:
The problem is in incorrect tun.c. Developers has backported a number of patches to tun.c to work with pseudo-socket object in tun.c, so the problem came with commt https://github.com/wl500g/wl500g/commit/d2195cd85cfc73b90a796911288846098f4126e3 I've revert this changes and build firmware with old version of tun.c. Everithing works like a charm. Download here:
https://drive.google.com/folderview?id=0B3siHuUeUYRBbk5CUXdwZmJLV0E&usp=sharing

ondrej-bartos, 2016-08-11:
this sounds really great, iI will give a try. which of these packages is compiled for WL500gP-v2 ?

dvglab, 2016-08-11:
I compiled firmware for RT-N16, RT-N12, RT-N10, RT-N15U, and firmware for WL500W and WL500gp-V2 will be compiled in next 2-3 hours

dvglab, 2016-08-11:
So everything is in place. If someone wants firmware with fixed tun.c for another router place your requests here.

themiron, 2016-08-11:
@dvglab hi, thnanks for effort!
may I know if tun or tap not working? or both?

dvglab, 2016-08-11:
Good day **themiron**, glad to see you!
At least tun interface do not work, there are at least 2 problems, first of all at my environment new tun.c  breaks shutdown tun interface anl lock the openvpn PID, kill -KILL don't work, error messages looks like in first message of the thread, the second I cant access any services at server vpn ip address

Some additional thoughts is here http://wl500g.info/showthread.php?17136-%CD%EE%E2%E0%FF-%EF%F0%EE%F8%E8%E2%EA%E0-1-9-2-7-d-rtn-%EE%F2-%FD%ED%F2%F3%E7%E8%E0%F1%F2%EE%E2-%E8-%CE%EB%E5%E3%E0/page390

themiron, 2016-08-11:
@dvglab I'll take a look. if no better solution, I'll revert that commit

themiron, 2016-08-11:
@dvglab  I'll take a look. if no better solution, I'll revert that commit.
Could you kindly please disable fastnat and recheck with current sources?

dvglab, 2016-08-11:
@themiron corresponding to https://github.com/wl500g/wl500g/issues/451 and similar messages in wl500g.info forum I've made a conclusion that problem is not in fastnat. It seems to me that the same problems should be in openwrt project, but I didn't found something similar.  Maybe missed some other kernel patch to right work with pseudo-sockets. But the bug is reproducing with 100% repeatability. 
It's time to take gdb in hand and hack tun.c and the kernel code, but I don't know how to take the gdb in hand ))

So in any case I'll check it with disabled fastnat.

themiron, 2016-08-11:
@dvglab got it. any test results reragding fastnat deps would be highly appreciated

dvglab, 2016-08-11:
@themiron so, i've just tested it on original source code and here is the results

```
[openvpn@router root]$ nvram get productid
RT-N16
[openvpn@router]$ nvram get misc_fastnat_x
0
[openvpn@router opt]$ wget http://10.8.0.1/file
Connecting to 10.8.0.1 (10.8.0.1:80)
^C

//connecting stage is endless//

[openvpn@router opt]$ killall openvpn
[openvpn@router opt]$ killall openvpn
[openvpn@router opt]$ tail /tmp/syslog.log
Aug 11 16:11:57 openvpn[416]: Initialization Sequence Completed
Aug 11 16:13:46 openvpn[416]: event_wait : Interrupted system call (code=4)
Aug 11 16:13:46 openvpn[416]: /sbin/route del -net 10.8.0.1 netmask 255.255.255.255
Aug 11 16:13:46 openvpn[416]: Closing TUN/TAP interface
Aug 11 16:13:46 openvpn[416]: /sbin/ifconfig tun0 0.0.0.0
Aug 11 16:13:56 kernel: unregister_netdevice: waiting for tun0 to become free. Usage count = 1

```
run in 2nd term **tcpdump -vvni tun0** 

`16:21:18.068177 IP (tos 0x0, ttl 64, id 43990, offset 0, flags [DF], proto TCP (6), length 60)
    10.8.0.6.54605 > 10.8.0.1.80: Flags [S], cksum 0x5274 (correct), seq 786993155, win 4380, options [mss 1460,sackOK,TS val 4294917077 ecr 0,nop,wscale 2], length 0

16:21:18.071118 IP (tos 0x0, ttl 64, id 1174, offset 0, flags [DF], proto TCP (6), length 60)
    10.8.0.1.80 > 10.8.0.6.54605: Flags [S.], cksum 0x753f (correct), seq 4036381123, ack 786993156, win 65535, options [mss 1368,nop,wscale 9,nop,nop,TS val 3187669307 ecr 4294917077], length 0

16:21:21.066184 IP (tos 0x0, ttl 64, id 4012, offset 0, flags [DF], proto TCP (6), length 60)
    10.8.0.1.80 > 10.8.0.6.54605: Flags [S.], cksum 0x753f (correct), seq 4036381123, ack 786993156, win 65535, options [mss 1368,nop,wscale 9,nop,nop,TS val 3187669307 ecr 4294917077], length 0

16:21:21.067660 IP (tos 0x0, ttl 64, id 43991, offset 0, flags [DF], proto TCP (6), length 60)
    10.8.0.6.54605 > 10.8.0.1.80: Flags [S], cksum 0x46bc (correct), seq 786993155, win 4380, options [mss 1460,sackOK,TS val 4294920077 ecr 0,nop,wscale 2], length 0

16:21:21.069667 IP (tos 0x0, ttl 64, id 4014, offset 0, flags [DF], proto TCP (6), length 60)
    10.8.0.1.80 > 10.8.0.6.54605: Flags [S.], cksum 0x6987 (correct), seq 4036381123, ack 786993156, win 65535, options [mss 1368,nop,wscale 9,nop,nop,TS val 3187669307 ecr 4294920077], length 0`

as we can see the remote endpoint sends initial packet but our side didn't see it and request handshake again and again. So the problem is 100% in network stack of tun.c with pseudo-sockets, as I said earlier probably we have lost some important patch for tun.c or for kernel, It's little be hard to me to hack the kernel, so maybe you can debug it or simply revert these changes as I did in my branch.

PS: @themiron, do you interesting in integrating openvpn into sources? I'm sure, I can support this part of build platform. It's not necessary to turn it on by default, but I can build and maintain firmware with openvpn integrated. I have about 250 pcs WL500W and RT-N16 routers in my environment and openvpn run on everyone and it compiled in fimware. What do you think? Otherwise I'll fork the sources and rework it to build with openvpn integrated.

themiron, 2016-08-11:
@dvglab thanks for investigation, have u reboot after seting fastnat to 0?
> do you interesting in integrating openvpn into sources?
not really, tinc-vpn.org is much better for me

dvglab, 2016-08-11:
Yep, I'm sure, I have rebooted it.

Also checked it:
```
[openvpn@rus_kir_283 root]$ cat /proc/sys/net/netfilter/nf_conntrack_fastnat
0
```

I can give you ssh access to the router to make you sure that all in places.   
So I'm going to fork the sources.

themiron, 2016-08-16:
tun generates wrong tcp/udp csums in tap mode.
fix will be available

themiron, 2016-08-19:
fixed with commit d251010

closed 2016-08-19 by themiron