wl500g/wl500g #480: miniupnpd is broken in r7441 and below

starter48, 2016-05-13:
Seems new **miniupnpd** is incompatible with current miniupnpd.conf and IP chains:
After start **miniupnpd**:
invalid option in file /etc/miniupnpd.conf line 3 : friendly_name=Unknown
invalid option in file /etc/miniupnpd.conf line 12 : lease_file=/tmp/upnp.leases

And after trying to map an UPnP port, in syslog.log:
May 13 18:59:52 miniupnpd[582]: addmasqueraderule() : chain UPNP-POSTROUTING not found
May 13 18:59:52 miniupnpd[582]: add_redirect_rule2(): addmasqueraderule returned -1
May 13 18:59:52 miniupnpd[582]: Failed to add NAT-PMP 1048 tcp->192.168.2.55:1048 'NAT-PMP 1048 tcp'
May 13 18:59:52 miniupnpd[582]: addmasqueraderule() : chain UPNP-POSTROUTING not found
May 13 18:59:52 miniupnpd[582]: add_redirect_rule2(): addmasqueraderule returned -1
May 13 18:59:52 miniupnpd[582]: Failed to add NAT-PMP 1048 udp->192.168.2.55:1048 'NAT-PMP 1048 udp'

_iptables -nL UPNP_ list is empty.

I don't know the first version number with broken miniupnpd, but **miniupnpd** from **r7235** works OK.

themiron, 2016-05-13:
chain error is fixed in r7443 with commit 102eddf00826d72054b9218a1fc6dd470cb10e33
as for invalid options, for what model are you building FW from sources?

starter48, 2016-05-13:
[RT-N16](http://asus.vectormm.net/rtn/7441/RT-N16-1.9.2.7-rtn-r7441M-gffcc5d2.trx)

themiron, 2016-05-13:
something is wrong with your nvram or build, friendly_name shouldn't be Unknown for RT-N16.

starter48, 2016-05-15:
I've upgraded to r7443 and reset setting. "addmasqueraderule" errors in syslog.log is gone, but "invalid option" errors is still there at console:
invalid option in file /etc/miniupnpd.conf line 3 : friendly_name=ASUS RT-N16
invalid option in file /etc/miniupnpd.conf line 12 : lease_file=/tmp/upnp.leases

If this chain unused, why don't just delete it and may be these options in miniupnpd.conf too?

lly-dev, 2016-05-15:
I can't reproduce the problem after theMIROn fix, even with from the image downloaded from http://asus.vectormm.net/rtn/
Looks like you have second local copy of miniupnpd binary compiled with some options disabled.

starter48, 2016-05-15:
I've installed http://asus.vectormm.net/rtn/7443/RT-N16-1.9.2.7-rtn-r7443M-g1060213.trx image,
and checked for another copy of miniupnpd :)
You may use the following steps to reproduce "invalid option" error:
_killall miniupnpd_
start _/usr/sbin/miniupnpd_ with full path to be sure :)
The errors will be printed immediately **at the console**, not in syslog.log

lly-dev, 2016-05-15:
Thanks, I miss that you mean console, not syslog.log. Problem was due to stale config.h content, fixed in commit 4a461257

closed 2016-05-15 by lly-dev