wl500g/wl500g #481: native dropbear DEFAULT_PATH isn't rich enough

starter48, 2016-05-15:
I've tried to start entware **rsync** via **native dropbear** shell, but unsuccessful:
**sh: rsync: not found**
However via **entware dropbear** that rsync can start without problems.
I've investigated problem and found the problem in DEFAULT_PATH variable (/gateway/dropbear/options.h). In the native dropbear it contains only "/usr/bin:/bin" string, whereas entware dropbear contains more rich DEFAULT_PATH: "/bin:/sbin:/usr/bin:/usr/sbin:/opt/bin:/opt/sbin".
You can check the default environment vars of dropbear using this command:
**ssh admin@my.router set**
(**set** command shall print all environment vars, including PATH='/usr/bin:/bin')
So the **native** dropbear can't start programs from /opt/... without explicit path in command line.
I suggest to borrow the value of DEFAULT_PATH from entware dropbear.

themiron, 2016-05-15:
optware/entware could be absent, so DEFAULT_PATH is not about to be changed.
dropbear starts shell with defaults, and shell have to set PATH to anything you want it to, upon starting /etc/profile gets executing, and it includes /usr/local/etc/profile and/or /opt/etc/profile executables if they're exist.
refer https://github.com/wl500g/wl500g/blob/master/gateway/others/profile#L6

closed 2016-05-15 by themiron

starter48, 2016-05-15:
It seems /etc/profile just ignored when start a command instead of interactive shell.
I mean if connect with "ssh admin@my.router" i'll get a shell with profile (and PATH) set.
But if start some command with "ssh admin@my.router command" i'll get a basic profile with DEFAULT_PATH.
I don't expect any side effects without entware and having "/opt/bin:/opt/sbin" in PATH since it is set in profile and does work :)
In other words, if entware can be both present and absent, why "/opt/bin:/opt/sbin" string should be present in /etc/profile and absent in dropbear's DEFAULT_PATH ?
I would be happy to start rsync with full path but it's hardcoded without path, so need to change some, and I thought you are less conservative :)

themiron, 2016-05-15:
what's the app with hardcoded rsync path?

lly-dev, 2016-05-15:
Since there is no ~/.ssh/environment equivalent in dropbear, it might be problems in some rare apps.

starter48, 2016-05-16:
2 themiron: It is **rsync** itself, but special version for Windows. It can start the server on demand via ssh, but expects rsync in PATH (hardcoded without explicit path). It's difficult to recompile and I believe it is dropbear's shortcoming :)
2 lly-dev: Can you provide an example of such "rare app" that can work with rich path in /etc/profile, but would fail if we extend dropbear's DEFAULT_PATH with "/opt/bin:/opt/sbin" at the tail?

themiron, 2016-05-16:
@starter48 ssh user can be unprivileged and therefore shouldn't have /sbin/, /*/sbin in its PATH.
dropber is simply relying on shell to use correct user's PATH after login.
that means, even if patch dropbear for /opt - /opt/sbin will not be there.
what about to add /usr/local/bin to default dropbear path, and it will be possible to have /usr/local/bin pointed to anyting (i.e with symlinking) else?

starter48, 2016-05-16:
Sorry, I don't understand this:
> that means, even if patch dropbear for /opt - /opt/sbin will not be there.
The default entware **rsync** executable folder is **/opt/bin**, not sbin
And then what about entware dropbear? It already contains hardcoded path "/bin:/sbin:/usr/bin:/usr/sbin:/opt/bin:/opt/sbin". And it does work. Do we have to drop it? :)
I've successfully tested your idea with symlink in /usr/local/... but I'm sure it is better to extend dropbear's DEFAULT_PATH :)

themiron, 2016-05-16:
obvilously entware dropbear has patched in a wrong way.

as for /opt in dropbear, it's not better. /opt may be accessible, but be locked in kernel (fs/usb/etc), under this circumstances any shell login attempt from some weird app relying on PATH will just hang instead of error out, and moreover it could lead to overall system hang/crash. so I'm not sure it's a good thing to do.

starter48, 2016-05-16:
I agree, but it is negligible, since we have PATH=/opt/sbin:/opt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin in /etc/profile also. So our firmware already affected.

themiron, 2016-05-16:
not really, we have /opt in PATH only after shell (/bin/sh) login.
after login it's up to user wether to call shell-builtin or fw-builtin apps (for example for solving usb hdd issues) or try to use external.
with remote executing there could be no such decide point, for example with your rsync.

starter48, 2016-05-16:
So do you agree to drop /opt/... folders from entware dropbear's path, and add /usr/local/... to native dropbear's path to make everything right? :)
Obeying your opinion, I think the best way to solve it is to add a new command line switch to dropbear to let user override dropbear's DEFAULT_PATH and thus get responsibility :)

themiron, 2016-05-16:
/opt prefix is default for entware, so it can't be and will not be dropped, /usr/local will be added to native
yes.. it could have different solutions, either to allow override PATH in dropbear,or force ash to use some environment file for non-interactive logins also, not decided yet

themiron, 2016-05-16:
as a temporary solution you could kill native dropbear and start entware's one in post-boot script

starter48, 2016-05-16:
I know. I've told about entware dropbear, so I've tried it of course.
Currently I've patched **rsync.exe** binary to use explicit path /opt/bin/rsync in a cost of overwriting version info (ugly patch).
I'll wait for native dropbear's DEFAULT_PATH update. Also I plan to blame dropbear author at the evening.

lly-dev, 2016-05-16:
First of all, you can extend path for specific command only, i.e.:
ssh admin@my.router "PATH=/opt/bin command" 

At second - Entware could be absent. In such case hardcoding of nonexistent path (/opt/bin) not a good solution and can add security hole, IMHO.

themiron, 2016-05-16:
@starter48 refer 13fdcf3b546858c53f61eae034e6ba6c9eb7b000 as half solution
full one will be to teach dropbear to use kind of ~/.ssh/environment file

@lly-dev he can't, cmd is hardcoded, and better would be use ssh user@host ". /etc/profile; cmd"

rozbo, 2017-06-21:
now,I meet this too. 
```shell
ssh user@host cmd
```
the path only /bin and /usr/bin

as u said ,it is possible

``` 
ssh user@host ". /etc/profile; cmd"
```

but I don't know why can't auto exec profile when client connection? that is well

themiron, 2017-12-02:
because it's supported only by shell only