wl500g/wl500g #482: pptp client

matrosoff, 2016-06-01:
Здравствуйте!

Подскажите, пожалуйста, как использовать на wl500gpv2 клиент pptp для подключения к pptp-серверу (для объединения двух удаленных локальных сетей, а не в рамках wan - не для доступа в интернет). Для доступа в интернет используется USB-модем.

Заранее большое спасибо.

TylerDurden7771, 2016-06-06:
я поднимал впн, но в прошивке 1.9.2.7-d-r2624. Сейчас не могу ее найти.

TylerDurden7771, 2016-06-06:
нашел. http://wl500g.googlecode.com/files/WL500W-1.9.2.7-d-r2624.trx

TylerDurden7771, 2016-06-06:
там 4 файла создается, в /usr/local/sbin/

themiron, 2016-06-06:
pptp client is builtin in any FW version,
you'll need to prepare pppd option with pptp module and corresponding ip-up, ip-down scripts

closed 2016-06-06 by themiron

matrosoff, 2016-06-06:
Огромное спасибо за ответ!

А если не затруднит - в какой именно момент правильнее вызывать pppd, дабы туннель создался после установления связи usb-модемом? Конфиг для pppd вот такой:

noauth refuse-eap
user 'user1'
password 'password1'
plugin pptp.so
pptp_server myserver.no-ip.org
nomppe-stateful -mppc mtu 1400
maxfail 0
persist
ipcp-accept-remote ipcp-accept-local noipdefault
defaultroute
usepeerdns
ktune
default-asyncmap nopcomp noaccomp
novj nobsdcomp nodeflate
lcp-echo-interval 10
lcp-echo-failure 6
unit 0

Правда его я нашел на форуме. Но нигде толком не смог найти ответы на свои вопросы:

1. Как сделать, что бы создавалось устройство, например ppp2 ?
2. Как реализовать роутинг только между клиентской локальной сетью и серверной локальной сетью, но что бы из локалки клиента был доступ в интернет через свое подключение, а не через серверное?
3. И еще - в скриптах ip-up, ip-down необходим вызов pppd, или что имелось в виде?

Заранее прошу прощения за ламерские вопросы.

TylerDurden7771, 2016-06-06:
у меня ip-up только без ip-down
вызов в  post-boot (сразу после загрузки) 
скрипт сделать исполняемым через chmod +x
#!/bin/sh
pppd file /tmp/local/sbin/options.wan1