wl500g/wl500g #489: ucd-snmp answer from wrong IP address

dvglab, 2016-09-06:
In case of runing openvpn and snmp request come through tun interface then ucd-snmp answers from incorrect IP address.

192.168.200.30 - host runing ping and snmpwalk command
10.4.49.1 - br0 IP address
10.8.5.74 - tun0 IP address

```
[admin@router etc]$ tcpdump -ni tun0
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 65535 bytes
05:18:00.102665 IP 192.168.200.30 > 10.4.49.1: ICMP echo request, id 5989, seq 3328, length 31
05:18:00.102817 IP 10.4.49.1 > 192.168.200.30: ICMP echo reply, id 5989, seq 3328, length 31
05:18:00.104684 IP 192.168.200.30.23144 > 10.4.49.1.161:  C=ro GetNextRequest(21)  .1.3
05:18:00.105840 IP 10.8.5.74.161 > 192.168.200.30.23144:  C=ro GetResponse(96)  .1.3.6.1.2.1.1.1.0="Linux router 2.6.22.21 #17 Fri Sep 2 13:24:01 MSK 2016 mips"

[admin@router etc]$ ifconfig 
br0       Link encap:Ethernet  HWaddr AC:22:0B:8D:8B:3E  
          inet addr:10.4.49.1  Bcast:10.4.49.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:40569 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16582 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:3279839 (3.1 MiB)  TX bytes:4041532 (3.8 MiB)

eth0      Link encap:Ethernet  HWaddr AC:22:0B:8D:8B:3E  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1127409 errors:0 dropped:0 overruns:0 frame:0
          TX packets:136199 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:108404858 (103.3 MiB)  TX bytes:20844010 (19.8 MiB)
          Interrupt:4 Base address:0x2000 

eth1      Link encap:Ethernet  HWaddr AC:22:0B:8D:8B:3E  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:3 Base address:0x1000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING MULTICAST  MTU:16436  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.8.5.74  P-t-P:10.8.5.73  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:69929 errors:0 dropped:0 overruns:0 frame:0
          TX packets:73738 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:4447598 (4.2 MiB)  TX bytes:6546696 (6.2 MiB)

vlan1     Link encap:Ethernet  HWaddr AC:22:0B:8D:8B:3E  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:40569 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16582 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:3442115 (3.2 MiB)  TX bytes:4107860 (3.9 MiB)

vlan2     Link encap:Ethernet  HWaddr AC:22:0B:8D:8B:3E  
          inet addr:195.14.X.X  Bcast:195.14.X.X  Mask:255.255.255.X
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1086840 errors:0 dropped:0 overruns:0 frame:0
          TX packets:119617 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:84669381 (80.7 MiB)  TX bytes:16736150 (15.9 MiB)

```

themiron, 2016-09-06:
provide iptables-save at the moment of observing replies from "incorrect" address

dvglab, 2016-09-06:
External IP addr have been masked. 
Also I've checked net-snmp and in case of use net-snmp instead of ucd-snmp everithing works lile a charm.

```
[admin@router etc]$ iptables-save
# Generated by iptables-save v1.4.3.2 on Tue Sep  6 07:24:13 2016
*nat
:PREROUTING ACCEPT [34450:2810140]
:POSTROUTING ACCEPT [6252:436777]
:OUTPUT ACCEPT [5597:391591]
:UPNP - [0:0]
:UPNP-POSTROUTING - [0:0]
:VSERVER - [0:0]
-A PREROUTING -d 195.14.XX.XX/32 -j VSERVER 
-A POSTROUTING ! -s 195.14.XX.XX/32 -o vlan2 -j MASQUERADE 
-A POSTROUTING -s 10.4.49.0/24 -d 10.4.49.0/24 -o br0 -j MASQUERADE 
COMMIT
# Completed on Tue Sep  6 07:24:13 2016
# Generated by iptables-save v1.4.3.2 on Tue Sep  6 07:24:13 2016
*mangle
:PREROUTING ACCEPT [219630:21926948]
:INPUT ACCEPT [208190:20722534]
:FORWARD ACCEPT [4125:310790]
:OUTPUT ACCEPT [191098:24721567]
:POSTROUTING ACCEPT [195224:25032685]
COMMIT
# Completed on Tue Sep  6 07:24:13 2016
# Generated by iptables-save v1.4.3.2 on Tue Sep  6 07:24:13 2016
*filter
:INPUT DROP [4877:370139]
:FORWARD DROP [0:0]
:OUTPUT ACCEPT [191098:24721567]
-A INPUT -i lo -j ACCEPT 
-A INPUT -i tun+ -j ACCEPT 
-A INPUT -i br0 -j ACCEPT 
-A INPUT -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT 
-A INPUT -s 195.14.XX.XX/23 -j ACCEPT 
-A INPUT -p icmp -j ACCEPT 
-A FORWARD -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT 
-A FORWARD -d 195.14.XX.XX/23 -j ACCEPT 
-A FORWARD -d 10.8.0.0/16 -j ACCEPT 
-A FORWARD -s 10.8.0.1/32 -j ACCEPT 
-A FORWARD -d 192.168.0.0/16 -j ACCEPT 
-A FORWARD -s 192.168.200.0/24 -j ACCEPT 
-A FORWARD -d 192.168.200.0/24 -j ACCEPT 
-A FORWARD -s 192.168.201.0/24 -j ACCEPT 
-A FORWARD -d 192.168.201.0/24 -j ACCEPT 
-A FORWARD -s 192.168.202.0/24 -j ACCEPT 
-A FORWARD -d 192.168.202.0/24 -j ACCEPT 
-A FORWARD -s 10.4.0.0/16 -i tun0 -j ACCEPT 
-A FORWARD -d 10.4.0.0/16 -o tun0 -j ACCEPT 
-A FORWARD -s 192.168.0.0/16 -i tun0 -j ACCEPT 
-A FORWARD -d 192.168.0.0/16 -o tun0 -j ACCEPT 
-A FORWARD -m iprange --src-range 10.4.49.150-10.4.49.160 -j ACCEPT 
-A OUTPUT -o lo -j ACCEPT 
COMMIT
# Completed on Tue Sep  6 07:24:13 2016

```

themiron, 2016-09-06:
ucd-snmp lacks of multihomed support.
fixed with 20ab482265169a70daa02927ebb39c8be5de3fa5 and 0881748d9a3568987ac621ae970c243c6d3d1649

closed 2016-09-06 by themiron