wl500g/wl500g #493: PPPoE connection fails with IPv6 set to automatic

isage, 2017-01-10:
r7510
It was working for some time, but now just disconnects whith ipv6 enabled.
```
Jan  1 03:00:02 syslogd started: BusyBox v1.24.2
Jan  1 03:00:03 kernel: klogd started: BusyBox v1.24.2 (2016-09-03 14:44:53 MSK)
Jan  1 03:00:03 kernel: Linux version 2.6.22.21 (root@localhost) (gcc version 4.6.4 (GCC) ) #2 Sat Sep 3 15:55:27 MSK 2016
Jan  1 03:00:03 kernel: CPU revision is: 00029006
Jan  1 03:00:03 kernel: Setting the PFC to its default value
Jan  1 03:00:03 kernel: Determined physical RAM map:
Jan  1 03:00:03 kernel:  memory: 02000000 @ 00000000 (usable)
Jan  1 03:00:03 kernel: Built 1 zonelists.  Total pages: 8128
Jan  1 03:00:03 kernel: Kernel command line: root=/dev/mtdblock2 rootfstype=squashfs noinitrd init=/sbin/init console=ttyS0,115200
Jan  1 03:00:03 kernel: Primary instruction cache 16kB, physically tagged, 2-way, linesize 16 bytes.
Jan  1 03:00:03 kernel: Primary data cache 16kB, 2-way, linesize 16 bytes.
Jan  1 03:00:03 kernel: PID hash table entries: 128 (order: -3, 512 bytes)
Jan  1 03:00:03 kernel: CPU: BCM4704 rev 9 at 264 MHz
Jan  1 03:00:03 kernel: Using 132.000 MHz high precision timer.
Jan  1 03:00:03 kernel: console [ttyS0] enabled
Jan  1 03:00:03 kernel: Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Jan  1 03:00:03 kernel: Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Jan  1 03:00:03 kernel: Memory: 28996k/32768k available (2710k kernel code, 3772k reserved, 527k data, 112k init, 0k highmem)
Jan  1 03:00:03 kernel: SLUB: Genslabs=7, HWalign=32, Order=0-1, MinObjects=4, CPUs=1, Nodes=1
Jan  1 03:00:03 kernel: Mount-cache hash table entries: 512
Jan  1 03:00:03 kernel: NET: Registered protocol family 16
Jan  1 03:00:03 kernel: PCI: Initializing host
Jan  1 03:00:03 kernel: PCI: Fixing up bus 0
Jan  1 03:00:03 kernel: PCI: Fixing up bridge
Jan  1 03:00:03 kernel: PCI: Fixing up bus 1
Jan  1 03:00:03 kernel: Time: MIPS clocksource has been installed.
Jan  1 03:00:03 kernel: NET: Registered protocol family 2
Jan  1 03:00:03 kernel: IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
Jan  1 03:00:03 kernel: TCP established hash table entries: 1024 (order: 1, 8192 bytes)
Jan  1 03:00:03 kernel: TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
Jan  1 03:00:03 kernel: TCP: Hash tables configured (established 1024 bind 1024)
Jan  1 03:00:03 kernel: TCP reno registered
Jan  1 03:00:03 kernel: squashfs: version 4.0 (2009/01/31) Phillip Lougher
Jan  1 03:00:03 kernel: io scheduler noop registered (default)
Jan  1 03:00:03 kernel: HDLC line discipline: version $Revision: 4.8 $, maxframe=4096
Jan  1 03:00:03 kernel: N_HDLC line discipline registered.
Jan  1 03:00:03 kernel: Serial: 8250/16550 driver $Revision: 1.90 $ 2 ports, IRQ sharing enabled
Jan  1 03:00:03 kernel: serial8250: ttyS0 at MMIO 0xb8000300 (irq = 3) is a 16550A
Jan  1 03:00:03 kernel: serial8250: ttyS1 at MMIO 0xb8000400 (irq = 3) is a 16550A
Jan  1 03:00:03 kernel: loop: module loaded
Jan  1 03:00:03 kernel: PPP generic driver version 2.4.2
Jan  1 03:00:03 kernel: PPP Deflate Compression module registered
Jan  1 03:00:03 kernel: PPP BSD Compression module registered
Jan  1 03:00:03 kernel: MPPE/MPPC encryption/compression module registered
Jan  1 03:00:03 kernel: Physically mapped flash: Found 1 x16 devices at 0x0 in 16-bit bank
Jan  1 03:00:03 kernel:  Amd/Fujitsu Extended Query Table at 0x0040
Jan  1 03:00:03 kernel: Physically mapped flash: CFI does not contain boot bank location. Assuming top.
Jan  1 03:00:03 kernel: number of CFI chips: 1
Jan  1 03:00:03 kernel: Flash device: 0x800000 at 0x1c000000
Jan  1 03:00:03 kernel: Physically mapped flash: squashfs filesystem found at block 1341
Jan  1 03:00:03 kernel: Creating 5 MTD partitions on "Physically mapped flash":
Jan  1 03:00:03 kernel: 0x00000000-0x00040000 : "boot"
Jan  1 03:00:03 kernel: 0x00040000-0x007f0000 : "linux"
Jan  1 03:00:03 kernel: 0x0014f4ec-0x007f0000 : "rootfs"
Jan  1 03:00:03 kernel: 0x007f0000-0x00800000 : "nvram"
Jan  1 03:00:03 kernel: 0x005b0000-0x007f0000 : "flashfs"
Jan  1 03:00:03 kernel: Found a 0MB  serial flash
Jan  1 03:00:03 kernel: sflash: found no supported devices
Jan  1 03:00:03 kernel: GACT probability NOT on
Jan  1 03:00:03 kernel: Mirror/redirect action on
Jan  1 03:00:03 kernel: u32 classifier
Jan  1 03:00:03 kernel:     Performance counters on
Jan  1 03:00:03 kernel:     input device check on 
Jan  1 03:00:03 kernel:     Actions configured 
Jan  1 03:00:03 kernel: Netfilter messages via NETLINK v0.30.
Jan  1 03:00:03 kernel: nf_conntrack version 0.5.0 (1024 buckets, 4096 max)
Jan  1 03:00:03 kernel: BCM fast NAT: INIT
[... 87 lines cut ...]
Jan  1 03:00:18 pppd[324]: Sent 0 bytes, received 0 bytes.
Jan  1 03:00:18 pppd[324]: Modem hangup
Jan  1 03:00:18 pppd[324]: Connection terminated.
Jan  1 03:00:18 pppd[324]: Sent PADT
Jan  1 03:00:18 dnsmasq[229]: read /etc/hosts - 3 addresses
Jan  1 03:00:18 dnsmasq-dhcp[229]: read /etc/ethers - 2 addresses
Jan  1 03:00:18 dnsmasq[229]: using nameserver 77.88.8.8#53
Jan  1 03:00:18 dnsmasq[229]: using nameserver 8.8.4.4#53
Jan  1 03:00:18 miniupnpd[225]: shutting down MiniUPnPd
Jan  1 03:00:18 miniupnpd[348]: HTTP listening on port 47877
Jan  1 03:00:18 miniupnpd[348]: Listening for NAT-PMP/PCP traffic on port 5351
Jan  1 03:00:18 PPPoE: connected to ISP
Jan  1 03:00:18 PPPoE: Disconnected
Jan  1 03:00:18 dnsmasq[229]: read /etc/hosts - 3 addresses
Jan  1 03:00:18 dnsmasq-dhcp[229]: read /etc/ethers - 2 addresses
Jan  1 03:00:18 dnsmasq[229]: using nameserver 77.88.8.8#53
Jan  1 03:00:18 dnsmasq[229]: using nameserver 8.8.4.4#53
Jan  1 03:00:27 zcip client: config IP : 169.254.82.67
Jan  1 03:00:27 dnsmasq[229]: read /etc/hosts - 3 addresses
Jan  1 03:00:27 dnsmasq-dhcp[229]: read /etc/ethers - 2 addresses
Jan  1 03:00:27 dnsmasq[229]: using nameserver 77.88.8.8#53
Jan  1 03:00:27 dnsmasq[229]: using nameserver 8.8.4.4#53
```

Note **Jan  1 03:00:17 pppd[324]: sif6addr: ioctl(SIOCADDRT): File exists (line 2606)**

themiron, 2017-01-10:
temporary turn on-demand off, if possible

isage, 2017-01-10:
What is "on-demand"? I can only set "IPv6 Connection Type" to Disabled or Automatic (because i don't know default gateway and netsize)

isage, 2017-01-10:
newermind, it looks like some problems on ISP side. Suddenly started working again.
However it's now spamming logs with
```
Jan 10 13:59:59 dhcp6c[363]: status code for identity association-505672: no addresses
Jan 10 14:00:00 dnsmasq[226]: read /etc/hosts - 3 addresses
Jan 10 14:00:00 dnsmasq-dhcp[226]: read /etc/ethers - 2 addresses
Jan 10 14:00:00 dnsmasq[226]: using nameserver 77.88.8.8#53
Jan 10 14:00:00 dnsmasq[226]: using nameserver 8.8.4.4#53
Jan 10 14:00:00 dhcp6c[363]: status code for identity association-505672: no addresses
Jan 10 14:00:00 dnsmasq[226]: read /etc/hosts - 3 addresses
Jan 10 14:00:00 dnsmasq-dhcp[226]: read /etc/ethers - 2 addresses
Jan 10 14:00:00 dnsmasq[226]: using nameserver 77.88.8.8#53
Jan 10 14:00:00 dnsmasq[226]: using nameserver 8.8.4.4#53
```

isage, 2017-01-10:
Sadly, this also renders router telnet or web-interface unusable after some time. Here's top output:
```
Mem: 17532K used, 11576K free, 0K shrd, 2376K buff, 5856K cached
CPU:   0% usr   1% sys   0% nic   0% idle   0% io   0% irq  98% sirq
Load average: 11.21 9.66 7.37 13/28 8530
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
    3     2 admin    SW<      0   0%  85% [ksoftirqd/0]
    4     2 admin    SW<      0   0%   8% [events/0]
  379     1 admin    R      872   3%   2% miniupnpd
  229     1 nobody   R      992   3%   2% dnsmasq
  221     1 admin    R      988   3%   1% httpd vlan1
  332     1 admin    R     1412   5%   1% watchdog
 8530  8523 admin    R     1400   5%   1% top
  324     1 admin    S     1504   5%   0% pppd file /tmp/ppp/options.wan0
  215     1 admin    R     1028   4%   0% nas
  231     1 admin    R      896   3%   0% lld2d br0 eth1
  363     1 admin    S     1412   5%   0% /sbin/dhcp6c -D LL -v ppp0
    1     0 admin    S     1412   5%   0% /sbin/init
 8523   220 admin    S     1404   5%   0% -sh
  322     1 admin    S     1400   5%   0% /sbin/udhcpc -i vlan1 -p /var/run/udhcpc0.pid -b -O33 -O121 -O249
  220     1 admin    S     1400   5%   0% telnetd
  209     1 admin    S     1396   5%   0% syslogd -m 0 -O /tmp/syslog.log -S -D -l 7 -b 1
  212     1 admin    S     1396   5%   0% klogd
  321     1 admin    S     1396   5%   0% /sbin/zcip -p /var/run/zcip0.pid vlan1 /tmp/zcip.script
  213     1 admin    R      836   3%   0% eapd
  335     1 admin    R      780   3%   0% infosrv br0
   64     2 admin    RW       0   0%   0% [pdflush]
  116     2 admin    SW<      0   0%   0% [mtdblockd]
    5     2 admin    SW<      0   0%   0% [khelper]
   27     2 admin    SW<      0   0%   0% [kblockd/0]
   63     2 admin    SW       0   0%   0% [pdflush]
   65     2 admin    SW<      0   0%   0% [kswapd0]
  244     2 admin    SW<      0   0%   0% [khubd]
    2     0 admin    SW<      0   0%   0% [kthreadd]
```

themiron, 2017-01-10:
cpu load is expected because due no hw pppoe acceleration and slow wl500gp's cpu.
as for IPv6-related flood issue, use more recent beta FWs from http://asus.vectormm.net/rtn/, at the moment last is 7545.

closed 2017-01-10 by themiron

isage, 2017-01-10:
Problem is not with high cpu load, but with high LA. After 10-15 minutes after router startup it becomes so uresponsive that it takes 3-5 minutes just to open web-interface. And it happens only with ipv6 enabled.

I'll retry with latest beta.

isage, 2017-01-10:
Looks like latest beta works fine, thanks